jmichault/flash_cc2531 #2: error on page 0: byte 0x00 is 0x00 instead of expected 0x02

georgcampana, 2019-04-25:
Hi I have the same issue reported today by caisyew on the other project:

https://github.com/Koenkk/zigbee2mqtt/issues/1430#issuecomment-486674644

by flashing with a RPI a cc2531 stick I get an error during the verification for the first byte of the first page. It writes 0x00 instead of the 0x02 value taken from the Hex file.

Maybe this is a regression of your fix from yesterday.  Checking the code right now. The hex file seems to be read in the right way since I dumped the data from the Page[0] structure.
It could be that the offset is not set in the right way. Checking your code right now :)

But probably you're faster :)

jmichault, 2019-04-25:
Hello, i have made a workaround so that both tested files works.

Hex file is correctly loaded. The matter is in write phase. Sometimes the number of bytes wrote is not what is expected. I must investigate for a complete fix.

jmichault, 2019-04-25:
Fixed : we must write complete words to flash.

closed 2019-04-25 by jmichault