jmichault/flash_cc2531 #3: cc_write hangs

kianusch, 2019-04-29:
I'm trying to flash a cc2530 via an raspberry.
It worked twice ... but not any more ... success is not reproducable.
Most of my tries hang on "page 1"

   writing page 1/127.

I have to press CTRL+C - to abort the flashing.

The ID gets recognized correctly - and I can also use cc_read to dump the current firmware.  So the raspberry setup and GPIO pins seem to be okay.

Reseting the board, rebooting the raspberry, unhooking/attaching the power supply does not help.

I've tracked down the problem to - the part 

"// wait DMA end :"

... but don't how to fix the problem ... might be a timing problem.

I tried two different CC2530 boards - on of the cc2530 I'm using is a E18-MS1-PCB.

Any ideas?

jmichault, 2019-04-29:
Hello,
i have only flashed cc2531 on my raspberry Pi 3 Model B+.
Timing can be a problem.

The timing is controlled by the cc_delay function in CCDebugger.c
You can try to change the multiplier on line 177 (actually 50).

What is the model of your raspberry ?

Have you tried different hex files ?

kianusch, 2019-04-29:
I'm using a raspberry pi zero.  Yep I tried different hex files.

Anyway - I solved my flashing problem, by taking your code - and also cc2530prog (which didn't work for me at) - and mixing both projects. I replaced most of the lowlevel codes in cc2530prog with the wiringPi code from your project - and it works now,

One thing not working, is - I can only flash sequential, starting at memory position 0x000000 - but I did not manage to address memory position not in sequence - I can not skip pages.  But this is not that much of a problem, because flashing takes only a few seconds - even when writing to all  memory positions.

AS137430, 2019-05-01:
@kianusch: Can you publish your code for the cc2530prog + flash_cc2531 project? I'm waiting for my E18-MS1-PCB to be delivered. Thanks.

jmichault, 2019-05-01:
@kianusch: Glad that my code helped you.

kianusch, 2019-05-03:
> @kianusch: Can you publish your code for the cc2530prog + flash_cc2531 project? I'm waiting for my E18-MS1-PCB to be delivered. Thanks.

You can find my code here ...

https://github.com/kianusch/flashPiCC

AS137430, 2019-05-05:
@kianusch thanks. Will be trying your code soon

g27radio, 2019-05-07:
Good to know about the timing thing.  I ended up here because I was having a similar problem with my original Pi 3 which had a lot of other processes running on it.  So I tried it on a fresh Pi 3 B+ and it flashed and verified on the first try.

yozik04, 2019-06-27:
Also do not use 3.3V pin on Raspberry Pi. It may not provide enough power.

ajain-93, 2019-08-05:
I've tried flashing using this process twice now in a months time, and both times I've encountered this issue with getting stuck at `writing page   1/128`. 

````bash
pi@raspberrypi:~/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page   1/128.^C
pi@raspberrypi:~/flash_cc2531 $
````

The way I managed to make it work was to before writing to the chip I made sure to erase the chip.

````bash
pi@raspberrypi:~/flash_cc2531 $ ./cc_erase
ID = b524.
erase result = 00a2.
pi@raspberrypi:~/flash_cc2531 $  ./cc_write CC2531ZNP-Prod.hex
ID = b524.
reading line 15490.
file loaded (15497 lines read).
writing page 128/128.
verifying page 128/128.
flash OK.
pi@raspberrypi:~/flash_cc2531 $
````

jmichault, 2019-10-03:
Yes, that's how the firmware flash works. 
You must erase before writing.

closed 2019-10-03 by jmichault