jmichault/flash_cc2531 #8: cc2531 wrong ID

AdrienAA, 2019-08-11:
Hello,

I'm trying to flash my cc2531, but the ID returned after running "./cc_chipid" is "ID = 1fff ."
I don't know where could the problem be. I checked my wiring a hundred time an everything seems to be ok.

I have no idea what to do, can you help me please ?

imdos, 2019-09-03:
You have probably twisted your cable (from the USB-stick to the downloader cable). I have experienced this as well.

Place the red side of the cable either both next to 1 or the other side around.

AdrienAA, 2019-09-23:
Hello,

I finaly have time to come back on my Zigbee devices.

So, I tried everything (and also I have seen the problem of people twisting their cable, but I took care to put everything well. My problem remains the same. I even bought another CC2531 USB Dongle but nothing changes...
Do you have any idea of where I can investigate ?

Thanks for you help !

AdrienAA, 2019-09-24:
Well, I have tried eveything. I reinstalled WiringPi, I checked everything in the wiring, but I still get as an ID : ffff
Does anyone have an idea ? I'm using a Raspberry Pi 3B+ with the latest version of raspbian and no SSH connection.

imdos, 2019-09-24:
I've succesfully used it with a batch on a pi3b. I've had problems with my downloader cable though. After I got a new one it worked perfectly. Altough the cable seemed intact it didn't worked. 

Otherwise power could be an issue maybe? Do you power the USB-stick from the Pi itself and do you have an appropriate power supply?

imdos, 2019-09-24:
Complain to the vendor of the stick? I've had two "debug locked" sticks myself. 
Maybe you could try to deduce the problem in your neighbourhood with someone who has a working environment?

AdrienAA, 2019-09-25:
Did you find any solution to bypass the "debug lock" ?
I'm afraid I don't have anybody around me who can help me

imdos, 2019-09-26:
No; not yet. I'm going to try once more with unitool from TI. I claimed my "broken" sticks with a dispute on Aliexpress. 

P.S. I saw your cross post [here](https://github.com/Koenkk/zigbee2mqtt/issues/2024)

AdrienAA, 2019-09-26:
I will try to claim a broken stick.
Yes, I made a post on the official zigbee2mqtt git, to see if they could have any other information

imdos, 2019-09-27:
Did you use SmartRF (v1) (on Windows) or unitool from ti.com?

For some USB-sticks which are failing with flash_cc2531 ([see this topic](https://github.com/jmichault/flash_cc2531/issues/9)), I can just flash them using a CC-Debugger and SmartRF on Windows 10 perfectly fine.

AdrienAA, 2019-09-30:
Yes, I use SmartRF V1 on windows.
Is it possible to use it directly on raspbian ?

imdos, 2019-09-30:
No SmartRF only works on Windows AFAIK. But you have to use the CC-debugger and such as explained [here](https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html). 

I've had 2 batches of cc2531 chips; one which I could flash with flash_cc2531 and my rPi3 and the others only through Windows,  CC-debugger and SmartRF. Which I explained in my [earlier post](https://github.com/jmichault/flash_cc2531/issues/8#issuecomment-535911549) above.

fredrikgk, 2019-10-01:
Koenkk/zigbee2mqtt#2024 Impossible to flash CC2531

fredrikgk, 2019-10-01:
More details here: [#2024](https://github.com/Koenkk/zigbee2mqtt/issues/2024)

AdrienAA, 2019-10-02:
Hello !
I bought a new cable to connect the dongle to the CCdebugger some times ago. It just arrived and everything worked ! So apparently the problem was only due to the cable.
Thanks a lot for your help ! I'm still trying to figure out the next steps, but finally I can go forward :)

closed 2019-10-02 by AdrienAA