jmichault/flash_cc2531 #10: Stuck on verifying

marrold, 2019-09-11:
Hi, I am trying to flash a CC2531 with a Raspberry Pi 1 B+ V1.2, however it's getting stuck on "verifying page" :

```
./cc_chipid
  ID = b524.
```

```
./cc_erase
  ID = b524.
  erase result = 00a2.
```

```
./cc_write CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page   1/128.
```

marrold, 2019-09-11:
Appears to work fine on a Raspberry Pi 3:

```
root@raspberrypi:/opt/flash_cc2531# ./cc_write CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page 128/128.
 flash OK.
```

luisapto702, 2019-09-13:
Im using raspberry pi4 and I tried several times flash cc2531 and all times occured problems in verifying page

`error at 0x0, 0x0 instead of 0xff
verifying page ........`

someone help me?

imdos, 2019-09-26:
> Im using raspberry pi4 and I tried several times flash cc2531 and all times occured problems in verifying page
> 
> `error at 0x0, 0x0 instead of 0xff verifying page ........`
> 
> someone help me?

Try it with a USB(2.0) hub. There appear to be some problems with the Pi4 and it's USB3. I have read a post of problems with an Aeontec Z-wave stick and Pi4 which was solved with a hub.

jmichault, 2019-10-03:
@luisapto702 
Can you retry with last release. It contains some changes in timings.

ditek, 2019-10-08:
Related to this, verification after flashing the chip fails on RPi4 but succeeds on RPi3. I tried with the latest release.

jbrocks, 2019-10-10:
I am having the same issue on RPi 3. Are there any solutions or workarounds? Should the stick work when quitting at verification time?

jbrocks, 2019-10-10:
Okay found a quick solution for myself. I tried to flash using openhabian. Somehow this problem does not occur with my other SD card, with a clean Raspbian installation. Seems to be a software issue.

kornikkk, 2019-10-19:
I'm having the same issue on Raspberry Pi B+ and the latest version. cc_read also won't work. I checked out commit 65b94487c6a0ae38f57e1aa1a35d79129c815faa from Apr 25, 2019 and everything works perfectly. Something must be wrong after this commit...

wesselah, 2019-10-22:
I have excactly the same  issue verifing got stuck using the latest comment meaning using 65b9448 worked for me too.

hmoffatt, 2019-11-07:
I had a verification failure with the latest release on Pi3, but I ran it again and it worked ok.

    pi@raspberrypi:~/src/flash_cc2531 $ ./cc_erase 
      ID = b524.
      erase result = 00a2.
    pi@raspberrypi:~/src/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex
      ID = b524.
      reading line 15490.
      file loaded (15497 lines read).
    writing page 128/128.
    verifying page  91/128.
    error at 0x0, 0xe0 instead of 0x22
    verifying page 128/128.
     Errors found in 1 pages.
    pi@raspberrypi:~/src/flash_cc2531 $ ./cc_erase 
      ID = b524.
      erase result = 00a2.
    pi@raspberrypi:~/src/flash_cc2531 $ ./cc_write CC2531ZNP-Prod.hex
      ID = b524.
      reading line 15490.
      file loaded (15497 lines read).
    writing page 128/128.
    verifying page 128/128.
     flash OK.

jdtsmith, 2019-12-03:
Same issue here with an RPI Zero W and Raspbian Buster. Any idea what would cause it to hang on:

> verifying page   1/128.

jdtsmith, 2019-12-03:
Update: To avoid hanging on the "verifying page" during install on RPIZeroW Buster, I had to recompile with a "make". Then it works fine. Perhaps just remove the executables and have people compile their own version?

nistvan86, 2019-12-09:
Going back to https://github.com/jmichault/flash_cc2531/commit/65b94487c6a0ae38f57e1aa1a35d79129c815faa solved it for me. Pi Zero, the older non W version. Recompiling the head was not enough.

marrold, 2019-12-16:
Same issue with the Raspberry Pi 1 B+ V1.2 again. Reverted to 65b9448 as suggested and it's working.

eon123, 2019-12-18:
I'm not so lucky as other here.  I'm getting the attached output when using a RPI 1B + V1.2 Sourcing with the latest version of the tools,, i.e. it's stuck at verifying page 1.  I tried reverting to [65b9448](https://github.com/jmichault/flash_cc2531/commit/65b94487c6a0ae38f57e1aa1a35d79129c815faa) as mentioned above but that version does not recognise the device at all and only outputs 0000's as the ID.  Any thoughts anyone? I am going to try and source a RPi2 to test with.

[image]

JoseAlcerreca, 2020-01-14:
Had the same problem in Rpi 1 B+ V1.2. Compiled master (just execute `make`) and it's working.

Verification is super slow though. I read the whole process would take 3 minutes but it's more like 1 page/minute.

oscarlinux, 2020-02-03:
> Im using raspberry pi4 and I tried several times flash cc2531 and all times occured problems in verifying page
> 
> `error at 0x0, 0x0 instead of 0xff verifying page ........`
> 
> someone help me?

Hi, Where you able to make it work on rasberry PI 4? I keep getting id=ffff...   thanks!

JakeMannen, 2020-02-09:
I needed to use the "-m 300" timings flag on my Pi B+ for this to work. Otherwise it got stuck at "Verifying page"
Good work here by the way!

leventebalogh, 2020-02-28:
**How to use on Pi 3 Model B**

Just like @JakeMannen said as well before the `-m 300` flag was needed in order to make the commands work, otherwise they returned with `id=ffff`.

This is what worked for me:
```bash
# 1. Build (Optional)
$ make

# 2. Test
$ ./cc_chipid -m 300

# 3. Save the flash content
$ ./cc_read save.hex -m 300

# 4. Erase
$ ./cc_erase -m 300

# 5. Flash
./cc_write -m 300 CC2531ZNP-Pro.hex
```

After this it works like a charm, amazing work, thanks! 🙏

waltner, 2020-04-29:
> I needed to use the "-m 300" timings flag on my Pi B+ for this to work. Otherwise it got stuck at "Verifying page"

Same here on a _Raspberry Pi 2 Model B Rev 1.1_. Thanks for the hint @JakeMannen ! :+1:
BTW, verification took about 2-3 sec per page.

tdgeorge, 2020-12-31:
Rebuilding HEAD fixed the issue for me on a Pi 3B.

Stefan300381, 2021-02-20:
another confirm that -m 300 fixed my frozen verification on raspberry Pi 1 Model B

closed 2021-05-08 by stale[bot]

Strand1984, 2021-08-12:
another confirm that -m 300 fixed my frozen verification

Staxen, 2021-08-31:
Confirm that -m 300 worked for my verifying error too...

verifying page   1/128.
error at 0x0, 0xff instead of 0x2
verifying page   2/128.
error at 0x0, 0xff instead of 0x8
verifying page   3/128.
error at 0x0, 0xff instead of 0x12