jmichault/flash_cc2531 #13: flash cc2531 with raspberry PI4

oscarlinux, 2020-02-04:
Hi, just wondering if this would work on a raspberry PI4. I'm following the steps but cc_chipid returns ffff. Has anyone been able to flash the cc2531 with a PI4? What other steps are required?

Thanks,
Oscar

dgericson, 2020-02-05:
> Hi, just wondering if this would work on a raspberry PI4. I'm following the steps but cc_chipid returns ffff. Has anyone been able to flash the cc2531 with a PI4? What other steps are required?
> 
> Thanks,
> Oscar

Did you double check your wiring? That sounds like what your problem might be.

I am attempting to flash a cc2531 on a Pi4 as well. I can see the device, read it, and erase it, but cannot write to it.

oscarlinux, 2020-02-06:
I checked the wiring a couple of times but didn't find any issue. I even checked continuity and all looks  good :/

oscarlinux, 2020-02-06:
> Did you double check your wiring? That sounds like what your problem might be.
>
> I am attempting to flash a cc2531 on a Pi4 as well. I can see the device, read it, and erase it, but cannot write to it.

I checked the wiring a couple of times but didn't find any issue. I even checked continuity and all looks good :/

dgericson, 2020-02-10:
I don't know if was just coincidental or not. But I installed network-manager and purged modemmanager, and the flash procedure worked perfectly.

pepijnmm, 2020-03-13:
@oscarlinux Did it still not work?
It worked for me. I connected the cc2531 to a exernal power suply but for the rest I did nothing diffrent then the tutorial on: https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html#with-raspberry-pi-3min

MichaelTurnbull, 2020-04-27:
I had an issue where cc_chipid was returning 0000 on my Raspberry Pi 4B. If anybody ends up here like me looking for an answer, in my case the problem was that the raspbian preinstalled version of wiringpi needs to be updated for the 4B. Once I updated I had no more issues.

http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/

gtb08177, 2020-06-15:
I am suffering the same behaviour described by @oscarlinux .

Running Raspbian Lite v10 on a Raspberry Pi 4 B 2Gb model. I am only ever getting 0000 for the cc_chipid step - both with normal wiring using depont cables and then using the alternative pins.

I tried going back through previous checkouts and unsuccessful. 

NB. I have installed the latest wiring pi as well. Any help appreciated.

Vaypron, 2020-06-27:
Hey guys, 
I had the same problem with my Raspberry Pi 4.
For me the follwing answer to another issue solved it:
https://github.com/jmichault/flash_cc2531/issues/9#issuecomment-558128187

gtb08177, 2020-06-28:
Thanks for the heads up @Vaypron sadly did not resolve my issue. I am beginning to think it is my specific C2531 debugger- however i did buy 2 and from 2 different suppliers. In case anyone can help, here is a little bit more context around my setup. 

dmesg printout
`[ 4250.948247] usb 1-1.1: new full-speed USB device number 23 using xhci_hcd
[ 4251.085134] usb 1-1.1: New USB device found, idVendor=0451, idProduct=16ae, bcdDevice=88.05
[ 4251.085141] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4251.085147] usb 1-1.1: Product: CC2531 USB Dongle
[ 4251.085152] usb 1-1.1: Manufacturer: Texas Instruments`

executing cchipid with alternate pins
`pi@RyPi:~/zigbee/flash_cc2531 $ ./cc_chipid -r 8 -c 0 -d 2
  ID = 0000.`

executing cc_erase
`./cc_erase -r 8 -c 0 -d 2
  ID = 0000.
  erase result = 0000.`

attempting the flash anyway (hangs here - 10 minutes given at most)
`pi@RyPi:~/zigbee/flash_cc2531 $ ./cc_write -r 8 -c 0 -d 2 CC2531ZNP-Prod.hex
  ID = 0000.
  reading line 15490.
  file loaded (15497 lines read).
writing page   1/128.`

execute usb-devices for dongle info
`T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#= 24 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=32 #Cfgs=  1
P:  Vendor=0451 ProdID=16ae Rev=88.05
S:  Manufacturer=Texas Instruments
S:  Product=CC2531 USB Dongle
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)`

Just can't seem to get the starts to align......... as ever - any help appreciated.

jmichault, 2020-06-30:
Have you tried the -m option with different values ?
`./cc_chipid -m 200 -r 8 -c 0 -d 2`

K-Visscher, 2020-08-04:
> Have you tried the -m option with different values ?
> `./cc_chipid -m 200 -r 8 -c 0 -d 2`

```bash
./cc_chipid -m 200
./cc_read -m 200 back-up.hex
./cc_write -m 200 CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
  writing page 128/128.
  verifying page 128/128.
  flash OK.
```

Adding the -m option with a value of 200, worked for me using a RPi 4, thanks!

marcusds, 2020-12-04:
> I had an issue where cc_chipid was returning 0000 on my Raspberry Pi 4B. If anybody ends up here like me looking for an answer, in my case the problem was that the raspbian preinstalled version of [...]

For those on pi Zero, I could not get it working until I updated ot wiringpi 2.52. I then had to set the time multiplier to at least 75, but went higher just incase. (IE ./cc_chipid -m 80)

ravngr, 2020-12-18:
> I had an issue where cc_chipid was returning 0000 on my Raspberry Pi 4B. If anybody ends up here like me looking for an answer, in my case the problem was that the raspbian preinstalled version of [...]

This was my experience too. Raspbian stable installed 2.50 via apt for me and would alternate chip ID between 0000 and FFFF depending upon CLI options. Installing 2.52 fixed it and I was able to erase and flash successfully.

closed 2021-05-08 by stale[bot]