jmichault/flash_cc2531 #14: Flash CC2530 ?

snakuzzo, 2020-02-20:
Hi, is it possible to flash a CC2530?
Flashing CC2530 with debugger I need to use also 3.3V pin...in the tutorial for CC2531 3.3V is not used.

JaviElio, 2020-03-17:
I am also interested if it works with CC2530

vladi4k, 2020-05-19:
Also works with Webee CC2530 + CC2591. You may have to try several times to flash and use extern 3.3V VCC not the Rpi 3.3V.

renatothome, 2021-02-11:
Hi all! I was able to flash the CC2530+CC2591 using this program in **RPi Pi 3b+** both firmware's 1 coordinator and 2 routes, but now I`m unable to **re-flash** the board again in order to updated the firmware. It keeps showing "ffff" when I run ./cc_chipid. 

What I`ve tried:
- Run the steps using RPi Pi 3b+ (using several -m options "100", "200", "300")
- Run the steps using RPi Pi 3b+ and using an external 3.3v to power the board (using several -m options "100", "200", "300")
- Run the steps using RPi Pi 4b (8gb) (Updated wiringPi to 2.52, using several -m options "100", "200", "300")
- Run the steps using RPi Pi 4b (8gb) and using an external 3.3v to power the board. (using several -m options "100", "200", "300")

it keeps returning "ffff". 

Any clues what I can do differently to **re-flash** the board?

cheers!

swimhero, 2021-02-16:
i do it well , and  when i let it work, it can not find the ttyACMO with the USB port

closed 2021-05-08 by stale[bot]

Morphy99, 2021-05-20:
> Hi all! I was able to flash the CC2530+CC2591 using this program in **RPi Pi 3b+** both firmware's 1 coordinator and 2 routes, but now I`m unable to **re-flash** the board again in order to updated [...]

Did you find a solution to this? I was going to try this but I don't want to brick my CC2530.