jmichault/flash_cc2531 #16: issues compiling on odroid-c2

elyorkhakimov, 2020-05-18:
OS: 
```
root@odroid:~/flash_cc2531# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.4 LTS"
root@odroid:~/flash_cc2531#
```

Hardware:

```root@odroid:~/flash_cc2531# lscpu
Architecture:        aarch64
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           4
Vendor ID:           ARM
Model:               4
Model name:          Cortex-A53
Stepping:            r0p4
CPU max MHz:         1536.0000
CPU min MHz:         100.0000
BogoMIPS:            2.00
Flags:               fp asimd crc32
root@odroid:~/flash_cc2531#```

running with WiringPi https://wiki.odroid.com/odroid-c2/application_note/gpio/wiringpi#tab__odroid-c2

Added few things to make sure Makefile compiles (the pre-compiled binaries would always say "directory not found"):

```root@odroid:~/flash_cc2531# cat Makefile
LDLIBS=-lwiringPi -lcrypt -lm -lrt
CFLAGS=-g
LDFLAGS=-g

all: cc_chipid cc_read cc_write cc_erase

cc_erase : cc_erase.o CCDebugger.o
        gcc $(LDFLAGS) -o $@ $^ $(LDLIBS)

cc_write : cc_write.o CCDebugger.o
        gcc $(LDFLAGS) -o $@ $^ $(LDLIBS)

cc_read : cc_read.o CCDebugger.o
        gcc $(LDFLAGS) -o $@ $^ $(LDLIBS)

cc_chipid : cc_chipid.o CCDebugger.o
        gcc $(LDFLAGS) -o $@ $^ $(LDLIBS)

cc_chipid.o : cc_chipid.c CCDebugger.h
        gcc $(CFLAGS) -c $*.c

CCDebugger.o : CCDebugger.c CCDebugger.h
        gcc $(CFLAGS) -c $*.c
root@odroid:~/flash_cc2531#```

Running make yields some warnings.

```root@odroid:~/flash_cc2531#
root@odroid:~/flash_cc2531# make
gcc -g -c cc_chipid.c
gcc -g -c CCDebugger.c
CCDebugger.c: In function ‘cc_reset’:
CCDebugger.c:421:12: warning: large integer implicitly truncated to unsigned type [-Woverflow]
   cc_delay(500);
            ^~~
gcc -g -o cc_chipid cc_chipid.o CCDebugger.o -lwiringPi -lcrypt -lm -lrt
cc -g   -c -o cc_read.o cc_read.c
gcc -g -o cc_read cc_read.o CCDebugger.o -lwiringPi -lcrypt -lm -lrt
cc -g   -c -o cc_write.o cc_write.c
cc_write.c: In function ‘main’:
cc_write.c:289:48: warning: format ‘%d’ expects a matching ‘int’ argument [-Wformat=]
     if(line%10==0) { printf("\r  reading line %d.");fflush(stdout); }
                                               ~^
cc_write.c:298:31: warning: format ‘%x’ expects argument of type ‘unsigned int *’, but argument 3 has type ‘uint16_t * {aka short unsigned int *}’ [-Wformat=]
       if(!sscanf(buffer+9,"%04x",&ela)) { fprintf(stderr,"incorrect hex file (incorrect extended addr)\n"); exit(1); }
                            ~~~^  ~~~~
                            %04hx
cc_write.c:317:35: warning: format ‘%x’ expects argument of type ‘unsigned int *’, but argument 3 has type ‘uint8_t * {aka unsigned char *}’ [-Wformat=]
       if(!sscanf(buffer+9+2*i,"%02x",&data[i])) { fprintf(stderr,"incorrect hex file (incorrect data)\n"); exit(1); }
                                ~~~^  ~~~~~~~~
                                %02hhx
gcc -g -o cc_write cc_write.o CCDebugger.o -lwiringPi -lcrypt -lm -lrt
cc -g   -c -o cc_erase.o cc_erase.c
gcc -g -o cc_erase cc_erase.o CCDebugger.o -lwiringPi -lcrypt -lm -lrt
root@odroid:~/flash_cc2531#
```

Can these warnings be ignored?

Also, when running the binaries, getting some errors:

```
root@odroid:~/flash_cc2531# ./cc_chipid
warn : pinMode: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : pinMode: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : pinMode: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : pinMode: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
  ID = 0000.
warn : pinMode: Not available for pin 28.
warn : digitalWrite: Not available for pin 28.
warn : pinMode: Not available for pin 28.
root@odroid:~/flash_cc2531#
root@odroid:~/flash_cc2531#
```

Output of Wiringpi:

```root@odroid:~/flash_cc2531# gpio readall -a
 +------+-----+----------+------+---+----+---- Model  ODROID-C2 ----+----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | DS | PU/PD | Physical | PU/PD | DS | V | Mode |   Name   | wPi | GPIO |
 +------+-----+----------+------+---+----+-------+----++----+-------+----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |    |       |  1 || 2  |       |    |   |      | 5V       |     |      |
 |  205 |   8 |    SDA.1 |   IN | 1 |    | DSBLD |  3 || 4  |       |    |   |      | 5V       |     |      |
 |  206 |   9 |    SCL.1 |   IN | 1 |    | DSBLD |  5 || 6  |       |    |   |      | GND(0V)  |     |      |
 |  249 |   7 | GPIO.249 |   IN | 1 |    |   P/U |  7 || 8  |       |    |   |      | TxD1     | 15  |      |
 |      |     |  GND(0V) |      |   |    |       |  9 || 10 |       |    |   |      | RxD1     | 16  |      |
 |  247 |   0 | GPIO.247 |   IN | 1 |    |   P/U | 11 || 12 | P/U   |    | 1 | IN   | GPIO.238 | 1   | 238  |
 |  239 |   2 | GPIO.239 |   IN | 1 |    |   P/U | 13 || 14 |       |    |   |      | GND(0V)  |     |      |
 |  237 |   3 | GPIO.237 |   IN | 1 |    |   P/U | 15 || 16 | P/U   |    | 1 | IN   | GPIO.236 | 4   | 236  |
 |      |     |     3.3V |      |   |    |       | 17 || 18 | P/U   |    | 1 | IN   | GPIO.233 | 5   | 233  |
 |  235 |  12 | GPIO.235 |   IN | 1 |    |   P/U | 19 || 20 |       |    |   |      | GND(0V)  |     |      |
 |  232 |  13 | GPIO.232 |   IN | 1 |    |   P/U | 21 || 22 | P/U   |    | 1 | IN   | GPIO.231 | 6   | 231  |
 |  230 |  14 | GPIO.230 |   IN | 1 |    |   P/U | 23 || 24 | P/U   |    | 1 | IN   | GPIO.229 | 10  | 229  |
 |      |     |  GND(0V) |      |   |    |       | 25 || 26 | P/D   |    | 1 | OUT  | GPIO.225 | 11  | 225  |
 |      |  30 |    SDA.2 |      |   |    |       | 27 || 28 |       |    |   |      | SCL.2    | 31  |      |
 |  228 |  21 | GPIO.228 |   IN | 1 |    |   P/U | 29 || 30 |       |    |   |      | GND(0V)  |     |      |
 |  219 |  22 | GPIO.219 |   IN | 1 |    |   P/U | 31 || 32 | P/U   |    | 1 | IN   | GPIO.224 | 26  | 224  |
 |  234 |  23 | GPIO.234 |   IN | 0 |    |   P/D | 33 || 34 |       |    |   |      | GND(0V)  |     |      |
 |  214 |  24 | GPIO.214 |   IN | 1 |    | DSBLD | 35 || 36 | DSBLD |    | 1 | IN   | GPIO.218 | 27  | 218  |
 |      |  25 |    AIN.1 |      |   |    |       | 37 || 38 |       |    |   |      | 1V8      | 28  |      |
 |      |     |  GND(0V) |      |   |    |       | 39 || 40 |       |    |   |      | AIN.0    | 29  |      |
 +------+-----+----------+------+---+----+-------+----++----+-------+----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | DS | PU/PD | Physical | PU/PD | DS | V | Mode |   Name   | wPi | GPIO |
 +------+-----+----------+------+---+----+---- Model  ODROID-C2 ----+----+---+------+----------+-----+------+
root@odroid:~/flash_cc2531#
root@odroid:~/flash_cc2531#
```

elyorkhakimov, 2020-05-18:
I might have found the reason why I had messages about "Not available for pin 28" -> this is a default setting. Once I changed to different pins I can now see a clean output. I have yet to connect the pins to the actual board itself.

```
root@odroid:~/flash_cc2531# ./cc_chipid -c 21 -d 22 - r 24
  ID = 0000.

```

The question about those compile time warnings remain, and plus I have a new question: 
- on RPI3 the "V" column in `gpio readall` command yields "0" value for lots of pins. The default pins in flash_cc2531 binaries use those pins with "0" values.  I see that on Odroid-C2 many of pins have "1" value. Is this a problem for flashing the firmware ?

jmichault, 2020-05-19:
Hello,
i have corrected code to suppress this warnings.
My code has never been tested on 64 bits architecture, and on other hardware than raspberry, you are the first to test it.

"1" value should not be a problem, state of pins is forced at beginning.

You should try the "-m" option with different values. Higher values usually give slowest speed but better stability.

elyorkhakimov, 2020-05-19:
thanks for quick fix! I will test it out and report back shortly

elyorkhakimov, 2020-05-20:
Success! 
I pulled from master, re-built binaries, and then did the flash.

```
root@odroid:~/flash_cc2531# ./cc_erase -c 23 -d 22 -r 24
  ID = b524.
  erase result = 00a2.
root@odroid:~/flash_cc2531# ./cc_write -c 23 -d 22 -r 24 CC2531ZNP-Prod.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128.
verifying page 128/128.
 flash OK.
root@odroid:~/flash_cc2531#
```

jmichault, 2020-05-20:
Great. :+1:

closed 2020-05-20 by jmichault