jmichault/flash_cc2531 #18: Directory not found using ubuntu on RPI 4b?

MikeZMssl, 2020-07-24:
Good afternoon,

After giving the command ./cc_chipid it says it cannot find the file or directory 
~/flash_cc2531$  ./cc_chipid
-bash: ./cc_chipid: No such file or directory

I do have WiringPi installed (gpio version: 2.50)
Does it have to do that I am running on Ubuntu 64 bit version (20.04 LTS) ?

Many thanks for your help!

scaytrase, 2020-08-11:
Same problem here

MikeZMssl, 2020-08-11:
A New sd card with raspbian lite did it for me!

scaytrase, 2020-08-11:
I manager to install `libwiringpi-dev` and run `make` successfully. but got `0000.`. Checking my wires now

Tobix99, 2020-11-22:
@scaytrase did you get it to work?

scaytrase, 2020-11-23:
If I remember correctly I moved to other flashing method. No luck with ubuntu 64 bit. I used wemos d1 mini (esp8266) as a intermediate flasher and then as a host device later.

https://zigbee.blakadder.com/flashing_ccloader.html

Tobix99, 2020-11-23:
Ahh yes I couldn't got it working yesterday and used a Raspbian Lite Version like @MikeZMssl on another SD Card and this worked.

albertmonzo, 2021-04-11:
This is because the version included in the repo is 32 bit, you have to recompike with `make`. If running on ubuntu debian you will also have to sort out wiringPi, first by adding `iomem=relaxed` to `/boot/firmware/cmdline.txt` and then downloading this wiringPI fork and applying the [patch here](https://github.com/WiringPi/WiringPi/issues/108#issuecomment-817384407)

closed 2021-06-17 by stale[bot]

Bagunda, 2022-07-12:
Same problem. The file exists. But the bash writes about the absence.

[image]

simonro0, 2023-03-18:
I had the same problem with raspbian 64bit. I just opened this repo called make and afterwards I received ID = 0000.
Without the 3.3V pin and inserted into a USB Port it returned ID = b524. and everything else worked as described.

tswart-res, 2023-08-09:
> I had the same problem with raspbian 64bit. I just opened this repo called make and afterwards I received ID = 0000. Without the 3.3V pin and inserted into a USB Port it returned ID = b524. and everything else worked as described.

Thank you for this - 64bit on my pi zero w 2 and was stuck until you pointed out I had to run the make command first.