jmichault/flash_cc2531 #20: Flash of CC2531 hangs

jottr, 2020-10-14:
I am trying to flash the **CC2531** with the the Z-Stack-firmware ([`CC2531_DEFAULT_20190608`](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default)).

I have the **CC2531** connected to my RaspberryPI via GPIO-Pins (`Raspberry Pi Model B Rev 2`).

When trying to flash, `cc_write` hangs on the first page, see below:

```

./cc_chipid -c6 -r 10 -d 11 
  ID = b524.

./cc_erase  -c6 -r 10 -d 11 -m 600
  ID = b524.
  erase result = 00a2.

/cc_write CC2531ZNP-Prod.hex -c6 -r 10 -d 11 -m 600
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page   1/128.

```

I've played with the delay timing to no avail.

I am running all of the above inside via `SSH` but inside of `TMUX`.

Jaren15949, 2020-10-26:
I'm having the exactly same problem with the same pi board

jottr, 2020-11-24:
Is anyone here who can help?
@beikeland maybe?

jottr, 2020-12-01:
Switching from a rPi 4 to an old model B seems to have done the trick.
Got it to work on the older rPi.
Not sure how this can be explained. 
Anyways, it works now.

closed 2020-12-01 by jottr

ragusa87, 2021-01-28:
I have the same issue using an old raspberry (Raspberry Model B Revision 2). I was able to read and clear the firmware, but the writing hangs.

I did some test my changing the time base (`-m` flag) with no success.

ragusa87, 2021-01-28:
I retried using a Raspberry PI 3B+, Revision: 03 and it works fine.

g-gg, 2021-07-14:
Had the same problem on a older Raspberry Pi (hangs at 1/128) and fixed this issue by connecting the CC2531 to an external USB power supply. And voilà, it worked!