jmichault/flash_cc2531 #22: wiringPi is deprecated

leo60228, 2020-12-08:
http://wiringpi.com/wiringpi-deprecated/

The official download for wiringPi is also down indefinitely: https://git.drogon.net/

leo60228, 2020-12-08:
libgpiod is the officially supported kernel interface for GPIO.

jmichault, 2021-02-06:
wiringpi is always in raspbian repository.
The official download is up, to install it, do this : 
`wget https://project-downloads.drogon.net/wiringpi-latest.deb
https://project-downloads.drogon.net/wiringpi-latest.deb`
There is an active fork on github : https://github.com/WiringPi/WiringPi.

AFAIK There have been no announcements from the maintainers of raspbian for the removal of wiringpi from the official repositories.

wait and see...

closed 2021-02-06 by jmichault

leonfunnell, 2021-03-08:
WiringPi doesn't work on the Compute Module 4 / IO board.  "Oops - unable to determine board type... model: 20"

jmichault, 2021-03-09:
Have you tried what is suggested in the **[documentation](https://jmichault.github.io/flash_cc2531-dok/)** ? 

> on pi 4 you will need version 2.52 of wiringpi : [install wiringPi 2.52 of Gordon website](http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/)

leo60228, 2021-03-09:
The Compute Module 4 is not the Pi 4.

leo60228, 2021-03-09:
https://github.com/WiringPi/WiringPi master got support for the CM4 late last year, but as far as I can tell nobody's ever used or tested it (the PR comes with a disclaimer that it's untested).

leo60228, 2021-03-09:
> AFAIK There have been no announcements from the maintainers of raspbian for the removal of wiringpi from the official repositories.

https://github.com/RPi-Distro/repo/issues/214#issuecomment-734402368

> I've left it in the repo for legacy applications and the plan was drop it after Buster if the situation doesn't improve.

leonfunnell, 2021-03-10:
Ok - the following works on CM4.  This is the new place for WiringPi and it has a patch for CM4.
`git clone https://github.com/WiringPi/WiringPi.git
cd WiringPi
./build
`

adiroiban, 2022-09-15:
you also need to (re)build for 64bit rpi OS.

But all good with getting WiringPI over git and rebuild the whole thing.

Thanks!