jmichault/flash_cc2531 #24: is it compatible with orangepi pc plus arm7 with WiringOP?

kirill-shershen, 2021-04-06:
Hey,
I have installed orangepi pc plus with WiringOP
gpio readall
[image]
is it posible to flash my cc2531 with this set?

kirill-shershen, 2021-04-06:
yes, it works on orangepi with WiringOP. I've checked it.

stevoh6, 2021-04-06:
@kirill-shershen  could you please share link to wiringop repo/tutorial? I was trying to do it with orangepi, but I failed. So I went with esp option. Thanks.

kirill-shershen, 2021-04-08:
> @kirill-shershen could you please share link to wiringop repo/tutorial? I was trying to do it with orangepi, but I failed. So I went with esp option. Thanks.
I've used this repo https://github.com/zhaolei/WiringOP.git with branch "h3".  
**gpio readall** to check it.
When i downloaded **flash_cc2531** i've launch "make" and then **cc_chipid** and it show me **ID = b524**.
then connect pins like on this picture
[image]
RPI pin layout here very similar with orangepi pc plus.
then launch cc_write

stevoh6, 2021-04-13:
@kirill-shershen thank you very much :-) Now I know, where I made mistake :-)

closed 2021-04-26 by kirill-shershen