jmichault/flash_cc2531 #29: wiringPi deprecated

rumoldus, 2022-02-20:
Hi, this project depends on wiringPi but that is no longer available. It cannot be installed anymore. So this flash software also will not run.

closed 2022-04-28 by stale[bot]

nobodyguy, 2022-05-25:
@rumoldus Keep a track of my fork - https://github.com/nobodyguy/flash_cc2531
I removed wiringPi dependency and replaced it with a standard libgpiod.
I'm still waiting for my CC2530 boards to arrive, so give me a week to be able to test my changes and push it.

Caius-Bonus, 2023-11-07:
@nobodyguy , Thanks for porting this to libgpiod. This project might become difficult to use without compatibility with current raspberry pi os versions. 

This project is important to keep these Zigbee devices accessible to people without CC programmer and these cheap devices are important to allow for accessible development for zigbee devices. 

However, I tested it with a CC2531 on a RPI3B on Bookworm and with `./cc_chipid` I got: `ID = 0000` . I didn't look further into it yet, but it still seems to work with https://project-downloads.drogon.net/wiringpi-latest.deb on a RPI3B. Would you be open to debug this? If @jmichault is open to it, it would be great to merge this here.