jmichault/flash_cc2531 #32: writing page  65/128.verification error

jcdel09, 2022-07-21:
hello,
chip it'is correct : ID = b524.
reading files OK:  reading line 15490.
  file loaded (15497 lines read).
have everytime one error at step 65.
writing page  65/128.verification error

closed 2022-09-27 by stale[bot]

Jordanvdb, 2022-11-01:
@jcdel09 I face exactly the same issue, I also tried the previous version of cc_write where the verification takes place after everything has been written, but then I get 99 pages with an error. 
Are you using a Raspberry Pi4? Wondering if it could have anything to do with it.
Not sure if it is a coincidence but page 65 is the first page after half the dataset..

jcdel09, 2022-11-13:
@Jordanvdb, thank for your coe back.
I paused the project for the moment. I will take this one again this winter. 
I use a Raspberry Pi3+
?

Jordanvdb, 2023-03-19:
@jcdel09 any luck with getting it working? I was using a Pi4.

jcdel09, 2023-03-27:
Hello,
yes i erased before write. it's always the same.
I saw that my chip was a cc2531 F 128. I bought another USB key with a 256 and it worked.
Moreover I think it's not the programming which is in question, but erasing. After programming with error, I reread the program and I found that on the non-programming part, I found the code of the old program. my solution was to buy another key. good luck

sigmdel, 2024-01-29:
The TI USB Dongle Hardware Description specifies a CC2531F256 (page 10 of _CC2531 USB Hardware User's Guide_ (swru221a)). However, like jcdel09, I obtained a dongle with a CC2531F128 (Jan 15, 2024). Unfortunately, 128K of flash memory cannot accommodate 240K of firmware. Hence the 

  `writing page  65/128.verification error... Have you erased before write ?`

error.

There is nothing wrong with **flash_cc2531**. It was easy to restore the previously saved factory sniffer firmware with `cc_write` after the flash had been clobbered with half of `CC2531ZNP-Prod.hex`.

The chip ID of my dongle is `b522`.  [LordMyshkin claims](https://github.com/Koenkk/zigbee2mqtt/issues/741)  to have successfully flashed a USB dongle with that ID. Consequently `ID = b522` does not identify a 128K flash CC2531.

Below a session showing how I checked that  **flash_cc2531** worked correctly with the CC2531F128.

1. Save the factory firmware and then try to flash ZNP.
```
$ ./cc_chipid -r 8 -c 0 -d 2 -m 300
ID = b522.
$ ./cc_read -r 8 -c 0 -d 2 -m 300 factory_sniffer.hex
ID = b522.
reading 256k/256k
$ ./cc_erase -r 8 -c 0 -d 2 -m 300
  ID = b522.
  erase result = 00a2.
$ ./cc_write -r 8 -c 0 -d 2 -m 300 CC2531ZNP-Prod.hex
  ID = b522.
  reading line 15490.
  file loaded (15497 lines read).
writing page  65/128.verification error... Have you erased before write ?
```
2. Obtain size of firmware by examining the Intel hex file  `CC2531ZNP-Prod.hex`.
```
$ grep ":02" CC2531ZNP-Prod.hex 
...
:020000040002F8
:020000040003F7

Upper 16 bit of 32 bit address: 03

$ tail -4 CC2531ZNP-Prod.hex 
:10C7E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF59
:10C7F000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF49
:0400000500002D6367
:00000001FF

Last address of lower 16 bit address: C7F0 + 10 = C800h

Therefore 3C800h = 247808 bytes in file.
```
3. Check that "half" of  `CC2531ZNP-Prod.hex` was uploaded to the flash memory
```
zigpi@zigpi:~/flash_cc2531-master $ ./cc_read -r 8 -c 0 -d 2 -m 300 broken.hex
  ID = b522.
 reading 50k/256k^C  (stop because 50 k is more enough to check)

zigpi@zigpi:~/flash_cc2531-master $ diff broken.hex factory_sniffer.hex -y
:020000040000FA					          :020000040000FA
:10000000000412000002E792AF22E4002000A3F0F7		      |	:100000000200E1020674FFFFFFFFFFFFFFFFFFFF9B
:1000100022A312000003F0227F0C02000000F5086A		      |	:10003000FFFFFF021229FFFFFFFFFFFFFFFFFFFF90
:10002000E4F5090000210B227401F0000020E064D7		      |	:10004000FFFFFF021129FFFFFFFFFF020767FFFF0E
:100030000F2274000023227F040228002023E024E2		      |	:10005000FFFFFF0207C5FFFFFFFFFFFFFFFFFFFFDF
:10004000F5C394022000058412AB8C02004B002201		      |	:10008000FFFFFF0205D400000080FB1200F0B90062
:10005000C082C00200130C7440F0A3000050A374CF		      |	:10009000030200DBE479217816B800028004F70936
:1000600052F0A3000060901B10742800206046F03E		      |	:1000A000D8FCE4900200789B79028002F0A3D8FC8F
:10007000A3742200207046F0A3745E00207046F046		      |	:1000B000D9FA90039AAA82AB839000F3781F790152
```
4. Restore the factory sniffer firmware
```
zigpi@zigpi:~/flash_cc2531-master $ ./cc_erase -r 8 -c 0 -d 2 -m 300
  ID = b522.
  erase result = 00a2.
zigpi@zigpi:~/flash_cc2531-master $ ./cc_write -r 8 -c 0 -d 2 -m 300 factory_sniffer.hex
  ID = b522.
  reading line 530.
  file loaded (539 lines read).
writing page   5/  5.
 flash OK.
```